YosysHQ/oss-cad-suite-build #14: GHDL and ghdl-yosys-plugin support

nobodywasishere, 2021-09-23:
Given YosysHQ/fpga-toolchain will soon be deprecated, having this repo support GHDL and ghdl-yosys-plugin will make it a lot easier for those who use VHDL to transition over.

qarlosalberto, 2021-10-15:
It would be really nice!!

mmicko, 2021-10-18:
This has been addressed with https://github.com/YosysHQ/oss-cad-suite-build/commit/a92aa960dd00ef122f8909870060bbdab724021b for linux-x64. Cross compiling for other platforms is currently a bit of issue.

petersieg, 2021-12-09:
Ahh. ok. Was just in the move to open an issue,because ghdl and ghdl-yosys-plugin is missing in darwin tgz :-(
Hope that can be solved in the future.
Anyway thanks a lot for this package!!
Best, Peter

jockm, 2022-09-23:
@mmicko  So it has been about a year since this was originally filed, and I find myself a little confused.  GHDL still isn't in the darwin build (but is in the linux-x64 build), nor is it mentioned anywhere I can find in the docs, or included tool, etc.  Does this mean if I compile it myself on macOS it will compile and be included or not?  

If so could I volunteer to do that so it could be included in the releases.  It would be nice to get it included on more platforms

jeremyherbert, 2022-09-26:
Also, just to add to this, it seems that the linux build doesn't actually build as such, it is just using an old cached version or something? Here is the output from `ghdl --version`:

```
GHDL 2.0.0-dev (tarball) [Dunoon edition] open-tool-forge.nightly-20210921
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Note that the copyright date was changed on the 1st of march: https://github.com/ghdl/ghdl/commit/8185ed878df5054a9624af97ae4c73bbd3e906ef so the build is at least 6 months behind (but probably more like 1 year based on the date in the version output)

mmicko, 2022-09-26:
@jockm please do check if GHDL support for darwin-x64 is working fine for you. I am using nightly builds since it can not be cross compiled for this platform, and updating libraries to make it work.
I have tried doing simple synthesis of VHDL file and it worked just fine. Please not that (due to license difference) GHDL for yosys is compiled as plugin so it requires ` yosys -m ghdl ` to load it.

jockm, 2022-09-28:
@mmicko Yes it seems to be working fine

nobodywasishere, 2023-01-07:
If it helps, I found a gcc ada compiler for M1 macs, don't have a computer on-hand to test building ghdl with it though.
https://github.com/simonjwright/distributing-gcc/releases/tag/gcc-12.2.0-aarch64

mmicko, 2023-01-07:
@nobodywasishere Thanks, but here we are using cross compiling for all platforms, so it requires ada cross compiler for specific platform

jockm, 2023-02-06:
@mmicko 

> @nobodywasishere Thanks, but here we are using cross compiling for all platforms, so it requires ada cross compiler for specific platform

I am deeply appreciative of all your hard work, but this is an unsatisfying response.  Shouldn't you change your approach to building so that you don't ignore a major platform?