YosysHQ/oss-cad-suite-build #15: feature request: change asset naming scheme to match git tag

ekiwi, 2021-11-10:
First of all I wanted to say that I love this project. This is the easiest way to get `yosys`, `verilator` and all the formal engines for use in CI. We are making use of this in the [firrtl CI](https://github.com/chipsalliance/firrtl/blob/18b9a987552492928aa1199d8cf498fe561c2f03/.github/workflows/test.yml#L23) as well as the [chisel CI](https://github.com/chipsalliance/chisel3/blob/614551236186d35ff42ea9c90130a3b80646ba69/.github/workflows/test.yml#L27). Since everything is hosted on github, the installation is blazingly fast.

The one feature that would improve our experience is if we could easily define the version of the oss cad suite that we want to use as an environment variable (or github action variable). The problem here is that the release date appears in two different formats in the URL. If you could adjust the naming scheme of the archive somehow to match the git tag, that would make changing the release with a single variable a lot easier:

```.diff
- https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2021-09-23/oss-cad-suite-linux-x64-20210923.tgz
+ https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2021-09-23/oss-cad-suite-linux-x64-2021-09-23.tgz
```

With this change we can use a shell variable like this to download the release:
```
https://github.com/YosysHQ/oss-cad-suite-build/releases/download/${VERSION}/oss-cad-suite-linux-x64-${VERSION}.tgz
```

mmicko, 2021-11-10:
First would like to thank you for kind words. Also I am very glad that CHIPS Alliance is using OSS CAD Suite builds for development workflows.
Agree that naming is a bit problematic in this case, but tgz name was actual scheme we used for our internal builds.

Change in code is quite trivial, but would like to check with other team members if there are some objections or use cases where this could break someones flow. Will rise this question to the rest of the team tomorrow.

To make things easy on your side I would suggest using https://github.com/YosysHQ/setup-oss-cad-suite in your workflows instead. That way you have just one date parameter to setup exact version (and this will work with any past or future version, even if we change naming scheme). 2nd example from https://github.com/YosysHQ/setup-oss-cad-suite/blob/master/README.md#usage is thing you need.

ekiwi, 2021-11-10:
> To make things easy on your side I would suggest using https://github.com/YosysHQ/setup-oss-cad-suite in your workflows instead. That way you have just one date parameter to setup exact version (and [...]

Oh, this is exactly what I was looking for, but somehow I wasn't aware that you had this action! I will try moving our CI to using the official action.

closed 2021-11-10 by ekiwi

mmicko, 2021-11-10:
@ekiwi Great. If you experience any problem using it, or need some specific improvement do not hesitate to raise an issue on that repository.

ekiwi, 2021-11-10:
> @ekiwi Great. If you experience any problem using it, or need some specific improvement do not hesitate to raise an issue on that repository.

OK. I will do that. Since most of our users are on Mac, it would be great if we can also run a mac CI.

ekiwi, 2021-11-10:
I am working on moving the firrtl and chisel CIs to use the action. They are all linux only, so I think it is going to be smooth sailing:
- https://github.com/chipsalliance/chisel3/pull/2243
- https://github.com/chipsalliance/firrtl/pull/2412
Thanks!