YosysHQ/oss-cad-suite-build #16: Builder.py Fails: FileNotFoundError: [Errno 2] No such file or directory: 'docker'

tortik92, 2021-11-14:
Hi! I am trying to build this distribution, but do have some troubles in building it.
After "./builder.py build" executed, I do have:
```
tortik92@LAPTOP-TQ37UJ6D:~/oss-cad-suite-build$ ./builder.py build
==> Loading default building rules ...
  -> Loading default ...
  -> Loading default-formal ...
  -> Loading ghdl ...
  ...
  -> Loading utils ...
  -> Loading verilator ...
  -> Loading yosys ...
  -> Loading xdot ...
  -> Loading graphviz ...
==> Validate building rules ...
==> Downloading sources ...
  -> [ghdl] Updating https://github.com/ghdl/ghdl ...
  -> [ghdl] Checkout origin/master ...
  -> [ghdl] Current revision e3c865e88bd12919a65e7230ce4b9772ea58d004 ...
  -> [python3] Updating https://github.com/python/cpython ...
  ...
  -> [nmigen-boards] Checkout origin/master ...
  -> [nmigen-boards] Current revision bd7fdd379d8b28f8b542f251a11ca28297e8fd6f ...
  -> [migen] Updating https://github.com/m-labs/migen ...
  -> [migen] Checkout origin/master ...
  -> [migen] Current revision 9a0be7a4210ff96043412539eb5388659b81831d ...
==> Building default for linux-x64 architecture ...
==> Step [ 1/50] building ghdl ...
  -> Remove old output dir ...
  -> Creating output dir ...
  -> Remove old build dir ...
  -> Creating build dir ...
  -> Copy 'ghdl' source to build dir ...
  -> Compiling ...
Traceback (most recent call last):
  File "./builder.py", line 85, in <module>
    cli(None)
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "./builder.py", line 38, in build
    buildCode(target, arch, nproc, force, dry, src, single, tar)
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 628, in buildCode
    code = executeBuild(target, arch, prefix, build_dir if not target.top_package else output_dir, output_dir, nproc, pack_sources)
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 431, in executeBuild
    return run_live(params, cwd=build_dir)
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 339, in run_live
    return asyncio.get_event_loop().run_until_complete(run_process(command, cwd, env))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 318, in run_process
    process = await asyncio.create_subprocess_exec(*command, cwd=cwd, env=env,
  File "/usr/lib/python3.8/asyncio/subprocess.py", line 236, in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 1630, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 197, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/lib/python3.8/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 789, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'docker'
```
Thank you for helping me!

mmicko, 2021-11-14:
@tortik92 you need to install docker first. For example if you are on Ubuntu you can follow https://docs.docker.com/engine/install/ubuntu/

tortik92, 2021-11-14:
Oh, I thought, that sudo apt-get docker would solve it...
Will try again to build it

> 
> 
> @tortik92 you need to install docker first. For example if you are on Ubuntu you can follow https://docs.docker.com/engine/install/ubuntu/

tortik92, 2021-11-14:
Well, now it is installed, but is not able to make a connection...

```
docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.
See 'docker run --help'.
==> ERROR : Script returned error code 125.
```
Do you possibly no why?

tortik92, 2021-11-14:
New error is here:

```

  -> [migen] Current revision 9a0be7a4210ff96043412539eb5388659b81831d ...
==> Building default for linux-x64 architecture ...
==> Step [ 1/50] building ghdl ...
  -> Remove old output dir ...
  -> Creating output dir ...
  -> Remove old build dir ...

==> ERROR : Error while deleting _builds/linux-x64/ghdl/ghdl.
```

closed 2021-11-15 by tortik92