YosysHQ/oss-cad-suite-build #17: Makefile:130 all Error 2

tortik92, 2021-11-15:
Hi! I do have troubles with building docker:
On the 12 of 50 step of buildng (nextpnr-ecp5) while compiling process on 92% of compiling this error:

```
[ 90%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o
[ 92%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/bitstream.cc.o
[ 92%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/cells.cc.o
{standard input}: Assembler messages:
{standard input}:45329: Warning: end of file not at end of a line; newline inserted
x86_64-linux-gnu-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/nextpnr-ecp5.dir/build.make:492: CMakeFiles/nextpnr-ecp5.dir/ecp5/arch_pybindings.cc.o] Error 1
make[2]: *** Deleting file 'CMakeFiles/nextpnr-ecp5.dir/ecp5/arch_pybindings.cc.o'
make[2]: *** Waiting for unfinished jobs....
/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/baseconfigs.cc: In function 'void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_45f(nextpnr_ecp5::ChipConfig&)':
/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/baseconfigs.cc:190:6: note: variable tracking size limit exceeded with '-fvar-tracking-assignments', retrying without
  190 | void config_empty_lfe5u_45f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/bitstream.cc: In function 'void nextpnr_ecp5::write_bitstream(nextpnr_ecp5::Context*, std::string, std::string)':
/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/bitstream.cc:585:6: note: variable tracking size limit exceeded with '-fvar-tracking-assignments', retrying without
  585 | void write_bitstream(Context *ctx, std::string base_config_file, std::string text_config_file)
      |      ^~~~~~~~~~~~~~~
/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/baseconfigs.cc: In function 'void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_85f(nextpnr_ecp5::ChipConfig&)':
/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/baseconfigs.cc:523:6: note: variable tracking size limit exceeded with '-fvar-tracking-assignments', retrying without
  523 | void config_empty_lfe5u_85f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [CMakeFiles/Makefile2:265: CMakeFiles/nextpnr-ecp5.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

==> ERROR : Script returned error code 2.
```

Used on WSL with Ubuntu 20.04.3 LTS

mmicko, 2021-11-15:
Hi @tortik92. WSL is not supported to be used as building platform. Please note that these same scripts/docker images pass fine on CI, so this build termination could be caused by lack of memory or some internal docker issue under WSL.

tortik92, 2021-11-15:
Ohhh, OK, thank you!

closed 2021-11-15 by tortik92