YosysHQ/oss-cad-suite-build #28: Could `nextpnr-gowin` support be added?

neilbalch, 2022-04-11:
One of the supported `nextpnr` architectures is `nextpnr-gowin`, but this must at present be compiled from source because it isn't distributed in this download. Could support for this `nextpnr` target be added?

mmicko, 2022-04-11:
Even it seems trivial, main issue packaging it is providing python packages and their deps (namely numpy which is huge one), since they are native libraries. This means that cross compilation is needed in order to get them packaged. As difference to other platforms this one requires python scripts to produce bitstream, not just to generate chip database.

mmicko, 2022-11-16:
This is now done for all supported platforms

closed 2022-11-16 by mmicko