YosysHQ/oss-cad-suite-build #32: Convert rbt bitstream to bin bitstream

liangkangnan, 2022-05-05:
Is there any tool can convert rbt format bitstream(generated by Lattice Radiant software) to bin format bitstream? Thank you.

gatecat, 2022-05-05:
No, though if you compared the two it shouldn't be too hard to write a simple python script to do it (although I'm mildly curious about your usecase...)

liangkangnan, 2022-05-05:
Thank you.
My usecase is: I have a ICE40 FPGA board，but it only support download rbt bitstream file. I want to use oss-cad tools to synth and route.

gatecat, 2022-05-05:
What board is this, out of curiosity? Also sounds like you are wanting to go in the other direction then; from bin to rbt?

liangkangnan, 2022-05-05:
Oh yes, from bin to rbt.
The board use lpc11u35(base on DAPLink) instead of ft2232 as a programmer.
The github repo is: https://github.com/EETree-git/ICE40_Training

liangkangnan, 2022-05-09:
I have solved the issue. Thank You.

closed 2022-05-09 by liangkangnan

tscmga, 2022-08-15:
> I have solved the issue. Thank You.

how ? i have the same problem

airskywater, 2022-10-03:
> 
@liangkangnan 
i have the same problem, and i try to write a python script to solve it :
https://github.com/airskywater/tillitis-key1/blob/main/hw/application_fpga/tools/bin2rbt/bin2rbt.py
but it does not work, could you  please help me?

airskywater, 2022-10-04:
> No, though if you compared the two it shouldn't be too hard to write a simple python script to do it (although I'm mildly curious about your usecase...)

@gatecat I have compared binary bitstream(which generated by Lattice Radiant software) and binary bitstream(which generated by icestorm/yosys/nextpnr), there are many differences, do you know why? 
I try to write a python script to solve it :
https://github.com/airskywater/tillitis-key1/blob/main/hw/application_fpga/tools/bin2rbt/bin2rbt.py.
It works for binary bitstream(which generated by Lattice Radiant software),
but it does not work for binary bitstream(which generated by icestorm/yosys/nextpnr), could you please help me?
Does it mean there are bugs in icestorm/yosys/nextpnr ?