YosysHQ/oss-cad-suite-build #41: Build failed

splinedrive, 2022-07-31:
oss-cad-suite-build $ ./builder.py build
oss-cad-suite-build $ ./builder.py build
oss-cad-suite-build $ ./builder.py build
oss-cad-suite-build $ ./builder.py build
+ cd prjoxide/libprjoxide
+ cargo install --no-track --path prjoxide --root /work/_outputs/linux-x64/prjoxide/yosyshq --target=x86_64-unknown-linux-gnu
  Installing prjoxide v0.1.0 (/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/prjoxide)
    Updating crates.io index
 Downloading crates ...
error: failed to compile `prjoxide v0.1.0 (/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/prjoxide)`, intermediate artifacts can be found at `/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/target`

Caused by:
  failed to download `hashbrown v0.12.3`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `/tmp/linux-x64/registry/src/github.com-1ecc6299db9ec823/hashbrown-0.12.3/Cargo.toml`

Caused by:
  feature `edition2021` is required

  this Cargo does not support nightly features, but if you
  switch to nightly channel you can add
  `cargo-features = ["edition2021"]` to enable this feature

==> ERROR : Script returned error code 101.
oss-cad-suite-build $ ./builder.py build
oss-cad-suite-build $ ./builder.py build
hd@t14:~/hacking/eda-tools/oss-cad-suite-build $ ./builder.py build
+ cd prjoxide/libprjoxide
+ cargo install --no-track --path prjoxide --root /work/_outputs/linux-x64/prjoxide/yosyshq --target=x86_64-unknown-linux-gnu
  Installing prjoxide v0.1.0 (/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/prjoxide)
    Updating crates.io index
 Downloading crates ...
error: failed to compile `prjoxide v0.1.0 (/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/prjoxide)`, intermediate artifacts can be found at `/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/target`

Caused by:
  failed to download `hashbrown v0.12.3`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `/tmp/linux-x64/registry/src/github.com-1ecc6299db9ec823/hashbrown-0.12.3/Cargo.toml`

Caused by:
  feature `edition2021` is required

  this Cargo does not support nightly features, but if you
  switch to nightly channel you can add
  `cargo-features = ["edition2021"]` to enable this feature

==> ERROR : Script returned error code 101.
hd@t14:~/hacking/eda-tools/oss-cad-suite-build $ ./builder.py build
Hi,

when I have done ./builder.py build I get this error, please help!

+ cd prjoxide/libprjoxide
+ cargo install --no-track --path prjoxide --root /work/_outputs/linux-x64/prjoxide/yosyshq --target=x86_64-unknown-linux-gnu
  Installing prjoxide v0.1.0 (/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/prjoxide)
    Updating crates.io index
 Downloading crates ...
error: failed to compile `prjoxide v0.1.0 (/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/prjoxide)`, intermediate artifacts can be found at `/work/_builds/linux-x64/prjoxide/prjoxide/libprjoxide/target`

Caused by:
  failed to download `hashbrown v0.12.3`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `/tmp/linux-x64/registry/src/github.com-1ecc6299db9ec823/hashbrown-0.12.3/Cargo.toml`

Caused by:
  feature `edition2021` is required

  this Cargo does not support nightly features, but if you
  switch to nightly channel you can add
  `cargo-features = ["edition2021"]` to enable this feature

==> ERROR : Script returned error code 101.

mmicko, 2022-07-31:
You are probably using old scripts, since in https://github.com/YosysHQ/oss-cad-suite-build/commit/b3d11b7379358626b0c4e5ede57651a0b09285b0 I have updated docker machines with latest Rust version for all, and build script is now using that version of docker images. Anyway all images are uploaded so no need for you to do manual build or anything.

closed 2022-07-31 by mmicko

splinedrive, 2022-07-31:
thank you for your quick help!

zjkyz8, 2023-04-06:
hi @mmicko , I use cross-darwin-arm64 based on master to build docker image, still have this error, is there any info i missed?