YosysHQ/oss-cad-suite-build #42: Issue in Readme - Litex install

NegativeEnergy, 2022-08-13:
Running Fedora 36 - when running the install instructions listed I am unable to get Litex to install, it has a conflict finding the Migen:
`AssertionError: Egg-link ./pyhdl/migen does not match installed location of migen (at /home/<USER>/Documents/Programs/OSS_FPGA/oss-cad-suite/lib/python3.8/site-packages/pyhdl/migen)` not sure if the issue is on Litex side or OSS side.

rcgoodfellow, 2022-09-04:
I hit this too. Was able to work around the issue by editing the file `<OSS_CAD_SUITE>/lib/python3.8/site-packages/migen.egg-link` which contained the relative links
```
./pyhdl/migen
.
```
and replacing them with the absolute link from the assertion error, e.g.
```
<ABSOLUTE_PATH>/<OSS_CAD_SUITE>/oss-cad-suite/lib/python3.8/site-packages/pyhdl/migen
```

DEFAULTTEXHAPb, 2022-11-09:
Are there any solutions how to create child virtual environment inside of (OSS CAD Suite) environment with own site-packages and read-only access to site-packages of the parent environment? It may be would solve this problem. Also it would be greate if using another python packages (like numpy / scipy / etc. ) for testing modules with cocotb.

georgantas, 2023-01-30:
Related: https://github.com/pypa/pip/issues/9452 ?