YosysHQ/oss-cad-suite-build #44: Can I use system apt-installed python3.10 to run related modules?

TurnOffNOD, 2022-09-15:
Hi, I am using linux mint 21 (a variant of ubuntu22.04). 

When I source <extracted_location>/oss-cad-suite/environment, it adds package-provided python3.8 in package to path ahead of system apt-installed python3 (python3.10). 

Is there any difference between package-provided python3 and official python3?

Another question is: is this software compatible with python3.10? Can I use system apt-installed python3 instead?

mmicko, 2022-09-15:
Hi. Please note that oss-cad-suite is self contained package, so it also contains python 3.8 and required libraries so it is able to run tools we provide. Note that some of tools (like nextpnr) are actually embedding python and they will use this 3.8 version even if your OS have different one. On the other hand there are tools using python only as interpreter but do require specific libraries to run (some of them have binary libraries as well) and all of that needs to be in sync in order to work.

Note that to use our tools you can just add oss-cad-suite/bin into your PATH and that would be enough. That way your default python will not be overridden but our tools will still use version of python we provide.

TurnOffNOD, 2022-09-18:
> Note that some of tools (like nextpnr) are actually embedding python and they will use this 3.8 version even if your OS have different one. 

Does "this 3.8 version" mean a modified 3.8 version compared to official 3.8 version?

>On the other hand there are tools using python only as interpreter but do require specific libraries to run (some of them have binary libraries as well) and all of that needs to be in sync in order to work.

Same question, does "specific libraries" mean a modified library, or just mean a specific version without modification?

> 
> Note that to use our tools you can just add oss-cad-suite/bin into your PATH and that would be enough. That way your default python will not be overridden but our tools will still use version of python we provide.

Some packages and scripts on my machine will also use this suite-contained python3.8 instead of system apt-installed python. So I want to make it clear whether it is modified.