YosysHQ/oss-cad-suite-build #46: Problem building natively on darwin-arm64

jockm, 2022-09-26:
After pulling the latest source and making sure I had all the build requirements, I attempted to build natively on dawwin-arm64 (`./builder.py build`), and got the following error:

```
Status: Downloaded newer image for yosyshq/cross-darwin-arm64:1.1

WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested

bash: /var/folders/b3/6wdzzgqj7rb8w1nmsndgyzwr0000gn/T/tmp212herc3: No such file or directory
```

I believe Docker is producing this error, and I will admit to being weaker there.  But it does make me wonder if I can even build natively if the docker image is linux at all.  

Any advice would be appreciated

mmicko, 2022-09-26:
Sorry but you can not build on ARM platform. Whole build process is cross compiling on x64 with prepared docker images, to be sure that exactly same libraries are used in build process and that there are no other dependencies.

closed 2022-09-26 by mmicko

jockm, 2022-09-26:
@mmicko Could all this be better documented?  This is very frustrating for someone who is just trying to find out if this suite will work for me and if I want to contribute in some way if it does.  The contributing section mentions nothing about the x86 build requirement, and there doesn't seem to be any other build documentation.