YosysHQ/oss-cad-suite-build #54: ghdl plugin with windows

Martoni, 2023-01-02:
Is there a plan to integrate ghdl yosys plugin to yosys for windows platforms ?

gfcwfzkm, 2023-06-09:
Would also be interested in it.
Is there any way to use ghdl with yosys?

Martoni, 2023-06-09:
> Is there any way to use ghdl with yosys?

Yes, in Windows, the good way to use ghdl with yosys is to use a [wsl](https://learn.microsoft.com/en-us/windows/wsl/install) distribution and follow [Linux](https://github.com/ghdl/ghdl-yosys-plugin) installation guide.

umarcor, 2023-06-09:
@Martoni, I wouldn't say that "use another OS" is the "good way" of doing something on one OS. It can be an "acceptable workaround", but I find it misleading to see it considered "good". In other words, WSL works the same as Docker, VirtualBox or any other virtualisation solution, because it's GNU/Linux, not Windows. More precisely, GHDL had problems on WSL1; it is reported to work on WSL2 because those are essentially containers. 

In order to have Windows binaries/executables that can be used on any Windows (Vista, 7, 8, 10, 11...) regardless of the version and OS virtualisation capabilities, the ghdl-yosys-plugin needs to be built into yosys (https://github.com/ghdl/ghdl-yosys-plugin#build-as-part-of-yosys-not-recommended), NOT as a module (https://github.com/ghdl/ghdl-yosys-plugin#build-as-a-module-shared-library). That's because, as far as I am aware, Yosys does not support loading shared libraries on Windows.

Package [mingw-w64-yosys](https://packages.msys2.org/base/mingw-w64-yosys) on MSYS2 includes ghdl-yosys-plugin already: https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-yosys/PKGBUILD#L46-L48. It's possible, although not straightforward, to create zipfiles from MSYS2 packages. See https://github.com/ghdl/ghdl/issues/1560.

Nevertheless, note that `ghdl synth` allows converting VHDL to Verilog without using Yosys. Then, Verilog sources can be fed into Yosys. Internally, GHDL does the same synthesis translations; the difference is the syntax of the output (Verilog or Yosys API). Several projects, such as Edalize or Litex, do use this two-step approach, instead of using ghdl as a plugin of yosys. See http://ghdl.github.io/ghdl/using/Synthesis.html#cmdoption-ghdl-out.

sirnails, 2025-06-01:
I use windows and write VHDL and i'd like to use MCY, I believe that I need to use the GHDL plugin for this. 

With this config.mcy:

```
[options]
size 100
tags COVERED UNCOVERED NOC

[script]
plugin -i ghdl
ghdl --std=08 counter.vhdl -e counter
hierarchy -top counter
proc; opt; memory; opt; fsm; opt

[files]
counter.vhdl
counter_tb.vhdl

[logic]
if result("vhdl_sim") == "FAIL":
    tag("COVERED")
    return

if result("eq_check") == "FAIL":
    tag("UNCOVERED")
    return

tag("NOC")

[report]
if tags("!NOC"):
    print("VHDL Mutation Coverage: %.2f%%" % (100.0*tags("COVERED")/tags("!NOC")))
    print("Total mutations: %d" % tags("!NOC"))
    print("Covered: %d" % tags("COVERED"))
    print("Uncovered: %d" % tags("UNCOVERED"))

[test vhdl_sim]
maxbatchsize 1
expect PASS FAIL
run bash $PRJDIR/run_vhdl_test.sh

[test eq_check]
maxbatchsize 1
expect PASS FAIL
run bash $PRJDIR/eq_check_vhdl.sh
```

I get this error:

```[OSS CAD Suite] C:\project\play-mutation>mcy init
==> Initialize database
  -> Reading configuration file.
  -> Checking database existance.
  -> Creating database directory.
  -> Creating design script file.
  -> Creating design RTL.
yosys -ql database/design.log database/design.ys
ERROR: Can't load module `./ghdl': "C:\OSS-CA~1\bin\..\share\yosys\plugins\ghdl.so": The specified module could not be found.
==> ERROR : Command 'yosys -ql database/design.log database/design.ys' returned non-zero return code 1.
```

I've been pulling my hair out trying to compile ghdl.so based on the documentation but failing. 
Is there any advice or help anyone can give please?

I'm going to try WSL and following the ubuntu guidance but would appreciate any help with getting this working in windows please!