YosysHQ/oss-cad-suite-build #64: GHDL is missing content from oss-cad-suite/include/ghdl/**

dlmiles, 2023-05-19:
After a regular GHDL autoconfig/configure/make/make install, the following files are installed:

```
# find /usr/local/include/ghdl/
/usr/local/include/ghdl/
/usr/local/include/ghdl/vhpi_user.h
/usr/local/include/ghdl/synth.h
/usr/local/include/ghdl/synth_gates.h
/usr/local/include/ghdl/libghw.h
/usr/local/include/ghdl/vpi_user.h
```

The files seems to be present in CI here:

https://github.com/YosysHQ/oss-cad-suite-build/actions/runs/5019966822/jobs/9001318938#step:18:243

But they are not present in the output tgz file:

```
tar -ztvf oss-cad-suite-linux-x64-20230519.tgz  | grep include/ghdl
```

The tree that is present is:

```
$ tar -ztvf oss-cad-suite-linux-x64-20230519.tgz  | grep include/py | grep "/$"
drwxr-xr-x root/root         0 2022-11-15 18:06 oss-cad-suite/lib/python3.8/site-packages/include/pygobject-3.0/
drwxr-xr-x root/root         0 2022-06-03 08:49 oss-cad-suite/include/python3.8/
drwxr-xr-x root/root         0 2022-06-03 08:49 oss-cad-suite/include/python3.8/internal/
drwxr-xr-x root/root         0 2022-06-03 08:49 oss-cad-suite/include/python3.8/cpython/
```

dlmiles, 2023-05-19:
Also to note that:

```
# ghdl --vpi-ldflags
--shared -L/opt/oss-cad-suite/libexec/lib -lghdlvpi -Wl,-rpath,/opt/oss-cad-suite/libexec/lib
# ghdl --vpi-cflags
-I/opt/oss-cad-suite/libexec/include/ghdl -fPIC

# ghdl --vpi-include-dir
/opt/oss-cad-suite/libexec/include/ghdl
# ghdl --vpi-library-dir
/opt/oss-cad-suite/libexec/lib

$ mkdir -pv /opt/oss-cad-suite/libexec/include/ghdl
mkdir: created directory '/opt/oss-cad-suite/include'
mkdir: created directory '/opt/oss-cad-suite/include/ghdl'
$ echo "I COPIED FILES FROM A WORKING INSTALL INTO HERE"

echo "I DID THIS FOR GOOD MEASURE"
$ mkdir -pv /opt/oss-cad-suite/libexec/lib
mkdir: created directory '/opt/oss-cad-suite/libexec/lib'
$ cd /opt/oss-cad-suite/libexec/lib
# ln -s ../../lib/libghdlvpi.so
# ln -s ../../lib/libghw.so
# ln -s ../../lib/libghdl.a
# ln -s ../../lib/libghdl-4_0_0_dev.so
# ln -s ../../lib/libghdl.link
```

Now it appears it maybe working.  Need to wait an report back on my unit testing simulation suite results.

Darksecond, 2023-07-20:
I'm running into this issue as well, the library and include paths GHDL reports seem to be incorrect, and the include headers seems to be missing entirely.

peterb12, 2025-02-07:
+1, I'm encountering this also.