YosysHQ/oss-cad-suite-build #69: Why the surelog plugin removed?

punzik, 2023-06-24:
Can you please tell me why they removed the surelog plugin? (commit: c3fe3a7686ed73e8eb6efe4b0a3b1243fafc9773)

sifferman, 2023-10-10:
I agree that Synlig would be a very useful re-addition to this repository. Is there any plan to add it back, @mmicko?

For the time being, fortunately [Synlig's releases](https://github.com/chipsalliance/synlig/releases/latest) are compatible with OSS CAD Suite's releases. Though there are some difficulties I ran into when installing that I am documenting here:

* glibc 2.35 is required by the current release. Therefore, Ubuntu 20.04 cannot be used, but Ubuntu 22.04 works (chipsalliance/synlig#1832)
* Place Synlig before OSS-CAD-Suite in your `PATH`. ex: `export PATH=$PATH:~/Utils/synlig/bin:~/Utils/oss-cad-suite/bin`

For example, here is a GitHub workflow that downloads both Synlig and OSS-CAD-Suite: <https://github.com/sifferman/eval-example/blob/0ae56d/.github/workflows/test.yml#L8-L20>

sifferman, 2023-11-11:
By the way, in case MacOS was the blocker, Synlig now compiles for MacOS: https://github.com/chipsalliance/synlig/pull/2133