YosysHQ/oss-cad-suite-build #70: AttributeError: module 'importlib.resources' has no attribute 'files'

multigcs, 2023-06-26:
gowin_pack -d GW1N-9C -o rio.fs rio_pnr.json
Traceback (most recent call last):
  File "/opt/oss-cad-suite/libexec/gowin_pack", line 33, in <module>
    sys.exit(load_entry_point('Apycula==0.8.2a1', 'console_scripts', 'gowin_pack')())
  File "/opt/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.8.2a1-py3.8.egg/apycula/gowin_pack.py", line 984, in main
AttributeError: module 'importlib.resources' has no attribute 'files'
make: *** [Makefile:16: rio.fs] Error 1

joyaledu, 2023-07-03:
Observing the same issue on Windows as well, `AttributeError: module 'importlib.resources' has no attribute 'files'`

multigcs, 2023-07-03:
file: lib/python3.8/site-packages/Apycula-0.8.2a2.dev1+g8e31050-py3.8.egg/apycula/gowin_pack.py

need to changes:
from:
`import importlib.resources`
...
`importlib.resources.files("apycula").joinpath(f"{device}.pickle"), 'rb') as f:`
to:
`from importlib_resources import files`
....
`files("apycula").joinpath(f"{device}.pickle"), 'rb') as f:`

multigcs, 2023-07-20:
i wrote a little patch script:
https://github.com/multigcs/LinuxCNC-RIO/blob/main/patching-oss-cad-suite.sh