YosysHQ/oss-cad-suite-build #72: yosys.exe is detected as a Trojan by Microsoft Defender

TraoreMorike, 2023-07-14:
Hello, 

As the title mentioned, yosys.exe from version : oss-cad-suite-windows-x64-20230714 is detected as :  Trojan:Win32/Bearfoos.A!ml by Microsoft Defender. 

I hope it's just a false positive. 

Best regards.

nakengelhardt, 2023-07-19:
We investigated and didn't find anything suspicious so far; I have also submitted the corresponding file to Microsoft for analysis. You may wish to [submit it also](https://www.microsoft.com/en-us/wdsi/filesubmission/) in case the file on your PC differs from what we get out of our github action.

MartinKavik, 2024-04-22:
Hello, `oss-cad-suite-windows-x64-20240422.exe` is automatically deleted by Defender immediately after downloading from https://github.com/YosysHQ/oss-cad-suite-build/releases:
[image]

jix, 2025-10-08:
@demon740 Note that the report mentions a "Heuristic analysis". Those are detection methods that try to detect new unknown threats using statistical methods such as machine learning or AI models. They are thus prone to have a high false positive rate. Nevertheless it is something to take serious and to investigate, so I checked the reported file with a wide array of AV engines: [VirusTotal Report](https://www.virustotal.com/gui/file/bb03a7f8af75c620d460abcdb71bc2e641bdd0c28d5f86a5f3659ef8a8cb2878/detection). You can confirm that this is the same file your report mentions by checking the MD5 hash on the "Details" tab.

Since no AV engine detected anything at all, and we have no indication of any issues besides the heuristic analysis report you provided, I am confident that the report you got is a false positive. Note that VirusTotal's reports includes Kaspersky's AV, but that Kaspersky may provide VirusTotal with a different configuration of their AV that excludes analysis that they know are prone to produce false positives.