YosysHQ/oss-cad-suite-build #81: Yosys crashes on LiteX build with `std::out_of_range`

josuah, 2023-08-24:
```
[...]
Info:     Created 2 LRAM_CORE cells from:
Info:              2x SP512K_MODE
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 64) > this->size() (which is 32)
build_lattice_crosslink_nx_evn.sh: line 4: 27089 Aborted                 nextpnr-nexus --json lattice_crosslink_nx_evn.json --pdc lattice_crosslink_nx_evn.pdc --fasm lattice_crosslink_nx_evn.fasm --device LIFCL-40-9BG400C --timing-allow-fail --seed 1
```

I will try to give a reproducer, for now I am bissecting it:

```
oss-cad-suite-2023-07-05.ok
oss-cad-suite-2023-08-13.err
oss-cad-suite-2023-08-22.err
oss-cad-suite-2023-08-23.err
```

More info about it tomorrow!

gatecat, 2023-08-24:
This was fixed yesterday in nextpnr https://github.com/YosysHQ/nextpnr/pull/1196, which should make it into today's build.

josuah, 2023-08-24:
Bad timing from me! I might have searched nextpnr a bit to early...
Thank you a lot for all the work on this!

closed 2023-08-24 by josuah