YosysHQ/oss-cad-suite-build #83: [Bug] `yosys.exe` doesn't output anything on Windows 10 Pro

pidgeon777, 2023-10-12:
I'm running this command:

```
yosys.exe xxx
```

The problem is that no matter which argument I use, `yosys` doesn't output anything.

I installed this Windows release with `scoop`:

https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-10-12

My OS is Windows 10 Pro.

mmicko, 2023-10-12:
Please note that in order to use tools from oss-cad-suite for Windows you need to have environment variables setup correctly, and to do that there are two ways:
1. start.bat that will automatically run command prompt and setup environment
2. opening command prompt and running environment.bat manually first

After these steps PATH will be setup correctly and some other variables as well making sure tools can run correctly.

Have tested exact build version you have pointed out on Windows 10 Pro and it is working as expected.

pidgeon777, 2023-10-12:
You were right, running `yosys.exe` after `start.bat` solved it.

Do you know which are the minimum environment variables to be set just to run `yosys.exe`?

mmicko, 2023-10-12:
You just need to set path to include bin and lib directories for yosys to work.

closed 2023-10-12 by mmicko