YosysHQ/oss-cad-suite-build #84: Verilator Binary Broken

sifferman, 2023-10-30:
Starting with release 2023-10-21, the Verilator binary gives the error "the coroutine header requires -fcoroutines". There is no error if I build from master.

## System Info

```plain
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
```

## Test

```systemverilog
module tb;
    initial begin
        #1;
        $finish();
    end
endmodule
```
```bash
verilator --binary tb.sv
```
### Full Error
```plain
make: Entering directory './obj_dir'
g++ -Os  -I.  -MMD -I./share/verilator/include -I./share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DVL_TIME_CONTEXT    -c -o verilated.o ./share/verilator/include/verilated.cpp
g++ -Os  -I.  -MMD -I./share/verilator/include -I./share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DVL_TIME_CONTEXT    -c -o verilated_timing.o ./share/verilator/include/verilated_timing.cpp
In file included from ./share/verilator/include/verilated_timing.h:45,
                 from ./share/verilator/include/verilated_timing.cpp:23:
/usr/include/c++/11/coroutine:334:2: error: #error "the coroutine header requires -fcoroutines"
  334 | #error "the coroutine header requires -fcoroutines"
      |  ^~~~~
In file included from ./share/verilator/include/verilated_timing.cpp:23:
./share/verilator/include/verilated_timing.h:98:10: error: ‘coroutine_handle’ in namespace ‘std’ does not name a template type
   98 |     std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
      |          ^~~~~~~~~~~~~~~~
./share/verilator/include/verilated_timing.h:112:44: error: expected ‘)’ before ‘<’ token
  112 |     VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
      |                      ~                     ^
      |                                            )
./share/verilator/include/verilated_timing.h:114:9: error: expected unqualified-id before ‘,’ token
  114 |         , m_process{process}
      |         ^
./share/verilator/include/verilated_timing.h:115:9: error: expected unqualified-id before ‘,’ token
  115 |         , m_fileline{fileline} {
      |         ^
./share/verilator/include/verilated_timing.h:115:32: error: expected unqualified-id before ‘{’ token
  115 |         , m_fileline{fileline} {
      |                                ^
./share/verilator/include/verilated_timing.h: In constructor ‘VlCoroutineHandle::VlCoroutineHandle(VlProcessRef)’:
./share/verilator/include/verilated_timing.h:108:11: error: class ‘VlCoroutineHandle’ does not have any field named ‘m_coro’
  108 |         : m_coro{nullptr}
      |           ^~~~~~
./share/verilator/include/verilated_timing.h: In constructor ‘VlCoroutineHandle::VlCoroutineHandle(VlCoroutineHandle&&)’:
./share/verilator/include/verilated_timing.h:122:11: error: class ‘VlCoroutineHandle’ does not have any field named ‘m_coro’
  122 |         : m_coro{std::exchange(moved.m_coro, nullptr)}
      |           ^~~~~~
./share/verilator/include/verilated_timing.h:122:38: error: ‘class VlCoroutineHandle’ has no member named ‘m_coro’
  122 |         : m_coro{std::exchange(moved.m_coro, nullptr)}
      |                                      ^~~~~~
In file included from ./share/verilator/include/verilated.h:42,
                 from ./share/verilator/include/verilated_timing.h:28,
                 from ./share/verilator/include/verilated_timing.cpp:23:
./share/verilator/include/verilated_timing.h: In destructor ‘VlCoroutineHandle::~VlCoroutineHandle()’:
./share/verilator/include/verilated_timing.h:129:25: error: ‘m_coro’ was not declared in this scope
  129 |         if (VL_UNLIKELY(m_coro)) {
      |                         ^~~~~~
./share/verilator/include/verilatedos.h:66:45: note: in definition of macro ‘VL_UNLIKELY’
[... 134 lines cut ...]
      |                                                            ^~~~

In file included from ./share/verilator/include/verilated.h:42,
                 from ./share/verilator/include/verilated_timing.h:28,
                 from ./share/verilator/include/verilated_timing.cpp:23:
./share/verilator/include/verilated_timing.cpp: In member function ‘void VlCoroutineHandle::resume()’:
./share/verilator/include/verilated_timing.cpp:31:19: error: ‘m_coro’ was not declared in this scope
   31 |     if (VL_LIKELY(m_coro)) {
      |                   ^~~~~~
./share/verilator/include/verilatedos.h:65:43: note: in definition of macro ‘VL_LIKELY’
   65 | # define VL_LIKELY(x) __builtin_expect(!!(x), 1)  // Prefer over C++20 [[likely]]
      |                                           ^
./share/verilator/include/verilated_timing.cpp: In destructor ‘VlCoroutine::VlPromise::~VlPromise()’:
./share/verilator/include/verilated_timing.cpp:217:9: error: ‘m_continuation’ was not declared in this scope
  217 |     if (m_continuation) m_continuation.destroy();
      |         ^~~~~~~~~~~~~~
./share/verilator/include/verilated_timing.cpp: At global scope:
./share/verilator/include/verilated_timing.cpp:220:6: error: ‘suspend_never’ in namespace ‘std’ does not name a type
  220 | std::suspend_never VlCoroutine::VlPromise::final_suspend() noexcept {
      |      ^~~~~~~~~~~~~
make: *** [./share/verilator/include/verilated.mk:278: verilated_timing.o] Error 1
make: Leaving directory './obj_dir'
%Error: make -C obj_dir -f Vtb.mk -j 1 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec verilator_bin --binary tb.sv
```

sifferman, 2023-11-03:
Could the issue be with [default/patches/verilated.mk.in](https://github.com/YosysHQ/oss-cad-suite-build/blob/master/default/patches/verilated.mk.in)? Although it looks like it's only being applied for Darwin releases, so I don't know why the Linux release would be affected

Maybe this commit was what broke things: [verilator/verilator 8b44a54b](https://github.com/verilator/verilator/commit/8b44a54bb225257690748acbe046935849053085)

sifferman, 2023-11-05:
Yes. I tried building Verilator but with [verilator/verilator/include/verilated.mk.in](https://github.com/verilator/verilator/blob/master/include/verilated.mk.in) replaced with [YosysHQ/oss-cad-suite-build/default/patches/verilated.mk.in](https://github.com/YosysHQ/oss-cad-suite-build/blob/master/default/patches/verilated.mk.in), and the error ("the coroutine header requires -fcoroutines") was replicated.

I believe removing the patch file will correct the error.

sifferman, 2023-11-06:
Alright, this is the commit that broke things:
> <https://github.com/verilator/verilator/commit/c7a0613c57827f0f9267fbfb0a064b5133a76678#diff-49473dca262eeab3b4a43002adb08b4db31020d190caaad1594b47f1d5daa810L374-R376>
> ```diff
> < _MY_CXX_CHECK_SET(CFG_CXXFLAGS_COROUTINES,-fcoroutines-ts)
> < _MY_CXX_CHECK_SET(CFG_CXXFLAGS_COROUTINES,-fcoroutines)
> < _MY_CXX_CHECK_SET(CFG_CXXFLAGS_COROUTINES,-fcoroutines-ts -Wno-deprecated-experimental-coroutine)
> ---
> > _MY_CXX_CHECK_IFELSE(
> >   -fcoroutines-ts,
> >   [CFG_CXXFLAGS_COROUTINES="-fcoroutines-ts"],
> >   [CFG_CXXFLAGS_COROUTINES="-fcoroutines"])
> ```

Verilator used to always include `-fcoroutines` by default, but now it only includes `-fcoroutines` if it is supported.
I believe that the error is because OSS CAD Suite builds Verilator with a C++ compiler that does not support coroutines. This can be corrected building Verilator with >=g++10

mmicko, 2023-11-06:
Thanks for looking into this issue. This is generaly hard to support on multiple linux distributions way we are building it, since it will always be affected by compiler on host machine as well. 

Anyway I am in process of updating base OS docker image, but it will take some time.

sifferman, 2023-11-23:
Would it be possible to just do a hotfix until the OS docker image is updated? I could throw together a patch if that would be helpful

mmicko, 2023-11-24:
@sifferman sure. please do. Will be happy to try it out

closed 2023-12-02 by mmicko