YosysHQ/oss-cad-suite-build #85: Could `nextpnr-xilinx` support be added?

sifferman, 2023-10-30:
Thanks for this project! It has saved me countless hours.

There is no release of `nextpnr-xilinx`, so it must be built from source. Could a binary for `nextpnr-xilinx` be added to this repository? <https://github.com/gatecat/nextpnr-xilinx>

If it's too difficult to support the UltraScale+ flow, then just the XC7 flow would still be a great.

Thanks!

disdi, 2023-12-20:
I would love to get support for [openXC7](https://github.com/openXC7) for xilinx here.

mmicko, 2024-01-02:
@sifferman nextpnr-xilinx is not in not in mainline nextpnr for a reason of not being "production ready" and mainly due to fact that chip databases are quite large. Building it and creating binaries part of oss-cad-suite-build is not that big effort, but we would also need to distribute chip databases but there are so many devices there and with large files per each it would take much much more space than currently.
So with current design of oss-cad-suite and how big chip databases are this is out of scope

closed 2024-01-02 by mmicko

jmi2k, 2025-06-28:
Does this extend to `himbaechel`'s support for Xilinx devices? I've spent the best part of a day trying to convince the CI to give me a build with support for it enabled, using [f4pga/prjxray-db](https://github.com/f4pga/prjxray-db) to gather info about a few devices. This adds around 20 MB to the final (Linux) tarball, and around 1 minute to the total run time.

If this sounds interesting, I can clean up my changes and make a PR.