YosysHQ/oss-cad-suite-build #88: Latest version fails to flash tang nano 9k

L10N37, 2023-11-23:
I used: 2023-03-23 
As per a youtube video guide, this fixed the problem.

https://www.youtube.com/watch?v=Y8koTqfXN3M

tmssngr, 2024-01-19:
Could someone from the YosysHQ team please comment on what causes the problems in newer builds? Are their plans to fix it? Do you need input from users?

gatecat, 2024-01-19:
Can you describe what the problem actually is, beyond a vague reference to a video? And what operating system you're using?

tmssngr, 2024-01-19:
IMHO the warning to use the old version of OSS-CAD-suite stems from this tutorial: https://learn.lushaylabs.com/getting-setup-with-the-tang-nano-9k/

> There are some synthesis issues in the latest version of OSS-CAD-Suite while they are going through a redesign of the OSS toolchain. It is recommended to download the 2023-02-10 version of OSS-CAD-Suite for best results: [https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-02-10](https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-02-10?ref=learn.lushaylabs.com)

I'm not experienced enough to verify whether their statement is correct. I've tried a tiny test verilog project with 2023-02-10 version, 2023-03-01 version and 2023-04-01 version. With the latter I had the most problems, but different ones than the warning states: failures to upload to the Tang Nano 9k. Probably because of the output of NUL-characters I can't copy paste the full log.
[image]

I've also tried with a build from today, but it looks like Lushay Lab's VSCode plugin is not compatible with the latest builds of OSS-CAD-Suite because it seems to miss some `bin\gowin_*.exe` files.