YosysHQ/oss-cad-suite-build #92: yosys.exe from OSS_CAD_SUITE does not work on Windows 10

marsohod4you, 2023-12-21:
I have downloaded  oss-cad-suite for windows from https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-12-21.
Then I start cmd and from cmd I run oss-cad-suite/environment.bat
Then I try to run yosys.

yosys.exe silently closes without any message without any file generation.
Generaly I want to build blinky example from apipula project:
**>yosys -D LEDS_NR=6 -p "read_verilog blinky.v; synth_gowin -json out/blinky.json"**
No output at all.

Starting like this gives nothing:
**>yosys --help** 

Linux version of yosys from oss-cad-suite generates files and drops many messages. It works on Ubuntu.

What I am doing wrong on WIndows?.

marsohod4you, 2023-12-23:
I have just tried another oss-cad-suite  https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2023-12-23/oss-cad-suite-windows-x64-20231223.exe

I do try in freshly installed Windows11 in VirtualBox.

The same issue - yosys.exe silently closes after start. Nothing heppens, no error and no any message..
I was trying to disable antivirus in Windows but nothing helps.

ayoung475, 2023-12-23:
I am also having this problem. No output at all when running yosys

mmicko, 2024-01-02:
Sorry for late reply. Can confirm this one. It is related to https://github.com/YosysHQ/yosys/issues/4068 builds before 1st of December works fine

closed 2024-01-07 by mmicko