YosysHQ/oss-cad-suite-build #107: MinGW, issue with VERILATOR_ROOT, verilator binary path

sylefeb, 2024-03-27:
**Platform:** MinGW64

**Problem:** The Verilator binary exists in two locations (`oss-cad-suite/share/verilator/bin/` and `oss-cad-suite/bin`), but using the one in `oss-cad-suite/bin` leads to an error `VERILATOR_ROOT is set to inconsistent path. Suggest leaving it unset.`. 

**Since when:** This seems to have been trigger on March 12 by commit 07326f2.

**Workaround:** Removing the binary from `oss-cad-suite/bin` and using the one in `oss-cad-suite/share/verilator/bin/` works.

davidar, 2024-03-30:
I'm experiencing the same issue on linux-x64

gainbrain, 2024-05-14:
Same here on darwin-x64. Apparently with commit 07326f28199f882c6bfadea5f502452990273121.

davekeeshan, 2024-06-10:
This is what they have said about it on the verilator github [#5047](https://github.com/verilator/verilator/issues/5047):

> VERILATOR_ROOT is intended to point to the directory that contains the bin and include subdirectories, like in a source tree. This was always the intention, even before the change, and is describe in the installation manual here: https://github.com/verilator/verilator/blob/master/docs/guide/install.rst#1-run-in-place-from-verilator_root
> 
> You are using installation method 2 from here: https://github.com/verilator/verilator/blob/master/docs/guide/install.rst#2-install-into-a-specific-prefix Albeit it's not explicitly said, this does not require setting VERILATOR_ROOT. The scripts know what it should be, and hence the error if you set it to something inconsistent. As the error suggests, you should leave VERILATOR_ROOT unset and just run the verilator executable, it should take care of everything else for you.
> 
> If you want to know what VERILATOR_ROOT is (though ideally you shouldn't have to care), run verilator --getenv VERILATOR_ROOT