YosysHQ/oss-cad-suite-build #112: OPENFPGALOADER_SOJ_DIR not set in environment script

exitrip, 2024-04-28:
proposed solution:
```
export OPENFPGALOADER_SOJ_DIR="$release_topdir_abs/share/openFPGALoader"
```

mmicko, 2024-04-29:
What is the use case for this one ? If you check in `bin` that variable is already set inside binary wrapper for `openFPGALoader` so it is enough for running this tool

exitrip, 2024-05-01:
It wasn't working inside of a Litex board module command until I added the export.  So I guess it's a Litex environment issue?
Specifically:
```
⦗OSS CAD Suite⦘ engin@engin-think:~$ CHIPDB=~/oss-cad-suite/chipdb python3 -m litex_boards.targets.colorlight_i9plus --build --flash --with-pmod-uart --with-spi-flash --toolchain=vivado
```