YosysHQ/oss-cad-suite-build #121: Killed:9 while trying to run GHDL Yosys plugin on darwin-arm64

consanii, 2024-08-29:
Pretty much as description.

I have fresh darwin-arm64 binaries from the release page (version 2024-08-29), and I have also run the `./activate` script to remove the quarantine flags. I source the environment as described on the readme with `source path/to/oss-cad-suite/environment.fish` on my fish shell and as soon as I try to run the Yosys GHDL plugin (with something like `yosys -m ghdl`) the process gets immediately killed.

If I use the darwin-x64 binaries everything works as expected.

My system is running on a Apple M1 Pro, on MacOS 14.6.1 .

ChristianJacobsen, 2024-09-01:
That's odd.

I can source both the `environment` and the `environment.fish` scripts from ZSH and Fish respectively. As far as I know I don't have any lingering `ghdl` modules stored anywhere on my machine from trying to configure this stack manually the first time.

Can you try `yosys -g -m ghdl`? Maybe there is some debug logging that could pinpoint this better. Alternatively could you try to open just `yosys`, and then - inside its shell - manually loading the plugin with `plugin -i ghdl`?

I'm just pulling these suggestions of out nowhere, btw. I'm about as "fresh" to this as it gets, but I did add the GHDL support for `darwin-arm64` - so I feel somewhat responsible.

consanii, 2024-09-02:
Hey, thanks for the reply.
I tried both commands, here is the result:
```bash
(OSS CAD Suite) ~/D/oss-cad-suite ❱ yosys -g -m ghdl

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.44+60 (git sha1 0fc5812dc, aarch64-apple-darwin21.4-clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)
fish: Job 1, 'yosys -g -m ghdl' terminated by signal SIGKILL (Forced quit)
(OSS CAD Suite) ~/D/oss-cad-suite 1.5s | 137 ❱ yosys

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.44+60 (git sha1 0fc5812dc, aarch64-apple-darwin21.4-clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)

yosys> plugin -i ghdl
fish: Job 1, 'yosys' terminated by signal SIGKILL (Forced quit)
(OSS CAD Suite) ~/D/oss-cad-suite 2.4s | 137 ❱ 
```

jtmaston, 2024-09-03:
Hi, chiming in with the same issue, M1 iMac, MacOS 14.6.1, Yosys 0.44+71.
Any luck getting it running?
Cheers!

jtmaston, 2024-09-03:
Hah! Found my issue. Apparently it was a signature caching issue, even though I ran the activate script.
For me, the fix was navigating to the oss-cad-suite/lib/ dir, the re-signing libghdl-5_0_0_dev.dylib, by running `codesign --sign - --force --preserve-metadata=entitlements,requirements,flags,runtime libghdl-5_0_0_dev.dylib`. All works fine after this.

I had yosys installed via Homebrew before, so it might have borked something in the signature caching inside MacOS.
Hope this helps you also!

consanii, 2024-09-04:
Nice catch @jtmaston, your solution fixed the problem for me as well!

For context, I have not installed yosys from Homebrew, but I do have multiple versions of the `oss-cad-suite` for darwin-x64. It's possible this is causing some interference.