YosysHQ/oss-cad-suite-build #122: Windows python3/pip3 ssl issue - cannot install litex

fanoush, 2024-09-08:
Hello,

I have extracted recent oss-cad-suite-build to C:\oss-cad-suite-build, run start.bat and included python3 cannot install any package due to ssl error. This is on Window 10

```
[OSS CAD Suite] C:\oss-cad-suite\litex>python3 -m pip install pyserial
DEPRECATION: Loading egg at c:\oss-cad-suite\lib\python3.11\site-packages\typing_extensions-4.12.2-py3.11.egg is deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use pip for package installation..
DEPRECATION: Loading egg at c:\oss-cad-suite\lib\python3.11\site-packages\yosys_mau-0.0.2-py3.11-mingw-x64.egg is deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use pip for package installation..
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pyserial/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pyserial/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pyserial/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pyserial/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pyserial/
Could not fetch URL https://pypi.org/simple/pyserial/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pyserial/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
ERROR: Could not find a version that satisfies the requirement pyserial (from versions: none)
ERROR: No matching distribution found for pyserial
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping

```

This also doesn't work with litex setup. As I need git for that I run Git for windows command prompt and the run environment.bat and then as per readme

https://github.com/YosysHQ/oss-cad-suite-build?tab=readme-ov-file#using-litex

the first `init` step works but then the setup fails on pip3 ssl issue too

```
[OSS CAD Suite] c:\oss-cad-suite\litex>python3 litex_setup.py install
          __   _ __      _  __
         / /  (_) /____ | |/_/
        / /__/ / __/ -_)>  <
       /____/_/\__/\__/_/|_|
     Build your hardware, easily!
          LiteX Setup utility.

[   0.003] LiteX Setup auto-update...
[   0.003] Installing Git repositories...
[   0.003] ------------------------------
[   0.003] Installing migen Git repository...
DEPRECATION: Loading egg at c:\oss-cad-suite\lib\python3.11\site-packages\typing_extensions-4.12.2-py3.11.egg is deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use pip for package installation..
DEPRECATION: Loading egg at c:\oss-cad-suite\lib\python3.11\site-packages\yosys_mau-0.0.2-py3.11-mingw-x64.egg is deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use pip for package installation..
Obtaining file:///C:/oss-cad-suite/litex/migen
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
      WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools/
      WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools/
      WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools/
      WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools/
      WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools/
      Could not fetch URL https://pypi.org/simple/setuptools/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/setuptools/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
      ERROR: Could not find a version that satisfies the requirement setuptools>=40.8.0 (from versions: none)
      ERROR: No matching distribution found for setuptools>=40.8.0
      Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
Traceback (most recent call last):
  File "c:\oss-cad-suite\litex\litex_setup.py", line 483, in <module>
    main()
  File "c:\oss-cad-suite\litex\litex_setup.py", line 463, in main
    litex_setup_install_repos(config=args.config, user_mode=args.user)
  File "c:\oss-cad-suite\litex\litex_setup.py", line 286, in litex_setup_install_repos
    subprocess.check_call("\"{python3}\" -m pip install --editable . {options}".format(
  File "c:\oss-cad-suite\lib\python3.11\subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"c:\oss-cad-suite\lib\python3.exe" -m pip install --editable . ' returned non-zero exit status 1.

[OSS CAD Suite] c:\oss-cad-suite\litex>
```

fanoush, 2024-09-08:
maybe this error helps?
```
[OSS CAD Suite] C:\oss-cad-suite>python3
Python 3.11.6 (tags/v3.11.6-dirty:8b6ee5ba3b, Feb  9 2024, 08:32:30) [GCC 13.2.1 20230728 (Fedora MinGW 13.2.1-5.fc39)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import ssl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\oss-cad-suite\lib\python3.11\ssl.py", line 258, in <module>
    from _ssl import enum_certificates, enum_crls
ImportError: cannot import name 'enum_certificates' from '_ssl' (C:\oss-cad-suite\lib\python3.11\lib-dynload\_ssl.cpython-311.dll)
>>>

```

matihadwork, 2026-09-10:
I can confirm that this still happens on 20260826 and on 20260909
`>>> import ssl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Software\oss-cad-suite\lib\python3.11\ssl.py", line 258, in <module>
    from _ssl import enum_certificates, enum_crls
ImportError: cannot import name 'enum_certificates' from '_ssl' (C:\Software\oss-cad-suite\lib\python3.11\lib-dynload\_ssl.cpython-311.dll) 
`