YosysHQ/oss-cad-suite-build #129: Nextpnr-ecp5  : dyld[10667]: symbol not found in flat namespace '___isPlatformVersionAtLeast'

elpekos, 2024-11-29:
dyld[10667]: symbol not found in flat namespace '___isPlatformVersionAtLeast'
BUG: on all latest releases of nextpnr-ecp5
please help !

thanks.

matiasilva, 2024-12-01:
I seem to be getting this as well, same OS

matiasilva, 2024-12-02:
Update: I've built nextpnr-himbaechel locally and it all works. I suspect this is a build issue as it affects other binaries too, like `python3`.

matiasilva, 2024-12-02:
I've dug deeper into this and I suspect that it's related to https://github.com/tpoechtrager/osxcross/issues/278.  Wondering if this SDK version is out of date? https://github.com/YosysHQ/oss-cad-suite-build/blob/main/docker/cross-darwin-arm64/Dockerfile.22#L41

mmicko, 2024-12-02:
@elpekos and @matiasilva thanks for reporting this, can confirm this issue, and it is caused by switch to newer SDK about half month ago. Looks like python and all binaries using python lib are affecting. Minimal supported SDK is set to be 11.0 so this should be fine and it links ok, so need to look deeper into this. Any help is appriciated.

mmicko, 2024-12-02:
Issue is now fixed and I have already rebuild todays release, so please do update.

matiasilva, 2024-12-02:
Excellent, thank you for the swift resolution!

elpekos, 2024-12-02:
I confirm : fixed. Thanks a lot !!!