YosysHQ/oss-cad-suite-build #133: Error while trying to use `show` with `slang` plugin

gmsanchez, 2024-12-18:
Hi,

I am helping the [APIO](https://github.com/FPGAwars/apio) developers to add SystemVerilog support in its ecosystem. Right now we are able to run some basic examples and we are getting some errors on anothers. I wonder if anyone could help or point me in some direction.

At the moment we are trying to run the [upduino32/blinky](https://github.com/FPGAwars/apio-examples/tree/master/examples/upduino31/blinky) example. After sourcing the OSS-CAD-SUITE I am running the following

```
yosys -m slang -p "synth_ice40 -top main -json _build/hardware.json" -q leds.v main.v oscilator.v
nextpnr-ice40 --up5k --package sg48 --json _build/hardware.json --asc _build/hardware.asc --report _build/hardware.pnr --pcf main.pcf -q
icepack _build/hardware.asc _build/hardware.bin
```

which works. Now we try to get an schematic of the circuit using

```
yosys -m slang -f slang -p "show -format dot -colors 1 -prefix _build/hardware main" -q leds.v main.v oscilator.v
Top level design units:
    leds

leds.v:37:3: error: unknown module 'SB_RGBA_DRV'
  SB_RGBA_DRV #(
  ^~~~~~~~~~~

Build failed: 1 error, 0 warnings
ERROR: Compilation failed
```

The `SB_RGBA_DRV` module is defined in the yosys library `tools-oss-cad-suite/share/yosys/ice40/cells_sim.v`, but is says it is not found. However, if I run the following I get no errors

```
yosys -f verilog -p "show -format dot -colors 1 -prefix _build/hardware main" -q leds.v main.v oscilator.v
dot -Tsvg _build/hardware.dot -o _build/hardware.svg
```

and the SVG file of the circuit is created.

Any help?

gmsanchez, 2024-12-19:
Hi @KrystalDelusion! Thanks for the quick response!

If I run the command you suggest, I get the same error. Apparently there is something in the way the `slang` plugin handles the `SB_RGBA_DRV` blackbox. A few hours ago I asked the same question in the yosys-slang repo: https://github.com/povik/yosys-slang/issues/72

In there, I got the commands to fail even before trying to use `show`. Check it out

```
yosys -m slang -p "synth_ice40 -top main -json _build/hardware.json" -q leds.v main.v oscilator.v
nextpnr-ice40 --up5k --package sg48 --json _build/hardware.json --asc _build/hardware.asc --report _build/hardware.pnr --pcf main.pcf -q
icepack _build/hardware.asc _build/hardware.bin
```
works, but using 

`yosys -m slang -f slang -p "synth_ice40 -top main -json _build/hardware.json" -q leds.v main.v oscilator.v`

results in the following error

```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.47+149 (git sha1 384c19119, clang++ 18.1.8 -fPIC -O3)

-- Parsing `leds.v' using frontend `slang' --

1. Executing SLANG frontend.
Top level design units:
    leds

leds.v:37:3: error: unknown module 'SB_RGBA_DRV'
  SB_RGBA_DRV #(
  ^~~~~~~~~~~

Build failed: 1 error, 0 warnings
ERROR: Compilation failed
```

I was able to get some SystemVerilog examples to work, but these examples are quite simple and do not use blackboxes. What would you suggest we should do? I think SystemVerilog support to the OSS-CAD-SUITE is great news and having that in APIO would be really nice.

Thanks

KrystalDelusion, 2024-12-19:
Ah, I think without the `-f slang` it's using the read_verilog front end instead of the slang front end (and the -m is just specifying to load the module). Definitely a problem for the slang repository then, glad you've already got an issue over there :)

gmsanchez, 2024-12-19:
Yes, indeed. After trying the commands that make the synthes I thought that was more of the `slang` plugin issue. I hope we can find a workaround for this.

BTW, I noticed that using yosys with SystemVerilog files and without specifying the frontend results in using the "-sv" frontend. If this frontend is available, why add the slang plugin? Which is the preferred way to handle SystemVerilog while using the OSS-CAD-SUITE?

KrystalDelusion, 2024-12-19:
The -sv frontend for the open source version uses the read_verilog command, which has very limited support for system verilog.  read_slang is newer and still in development, but [slang](https://github.com/MikePopoloski/slang) is much more fully featured and robust.  At this time, I would suggest trying the -sv frontend; if it works as expected then that is great, but if not then the -slang frontend may work with a bit of extra effort.

gmsanchez, 2024-12-19:
Thanks for the clarification.

I am really hyped about this after I found out about SystemVerilog support with the `slang` plugin and started trying to get some examples to work.

Right now all the code I tested are just basic modules, but they work as expected. I was able to simulate using gtkwave, impact on a [EDU-CIAA FPGA](https://www.proyecto-ciaa.com.ar/devwiki/doku.php%3Fid=desarrollo:edu-fpga.html) (which is a Lattice iCE40 HX4K) and get the graphs shown. That's a lot!

Since then, the people at APIO upgraded the OSS-CAD-SUITE in order to see if we can get APIO to support SystemVerilog too, which would be awesome.