YosysHQ/oss-cad-suite-build #143: Include relase version in README (or separate file)

secworks, 2025-04-02:
The name of the release build files include date information. But the extracted directory does not (which is good). But the contents in the dir does not include any information about what release it contains. This makes it harder to determine which release it contains.

As an improvement, could information about the release be added to either the README or possibly a separate file? For example VERSION or RELEASE?

mmicko, 2025-04-02:
Thanks for suggestion @secworks 
There are actually couple of ways to get information:
 1. Executing ` tabbyadm version` command 
 ```
OSS CAD Suite info

Product: oss-cad-suite
Architecture: linux-x64
Version: yyyymmdd
Package: oss-cad-suite-linux-x64-yyyymmdd
```
2.  Reading same data from `share/manifest.json`

Does using this solution works for you ?

secworks, 2025-04-02:
Almost, possibly. It depends on how that information or method is provided. 

If the README would include information about doing those things you mentioned to find out what version the installed dir contains it would be a step in the right direction.

But IMHO Having a file called RELEASE, VERSION or somesuch that display the release makes it easier for the user. 

It could be the release date and the digest for the release, as is prepended onto the releases file here when the release is done: https://github.com/YosysHQ/oss-cad-suite-build/releases

It could also be what you would get when running the 'tabbyadm version' command would show, which obviously is stored in the release when it is generated. Creating a file RELEASE should be possible to do automatically at the same time.

mmicko, 2025-04-02:
Fixed in https://github.com/YosysHQ/oss-cad-suite-build/commit/80695491cde35a08edbe8a4e608f77c9498061c6 
Have triggered build so in hour or so all builds will be updated.