YosysHQ/oss-cad-suite-build #148: WIndows: GHDL plugin for YoSys

sigmuha, 2025-04-07:
This is a feature request for a Windows build of the GHDL plugin from YoSys, as https://github.com/YosysHQ/oss-cad-suite-build#plugins only lists linux and macos builds. 

Is there a reason for not including windows here? 

Might seem like a duplicate of #146, but I forgot to mention windows, before it got closed.

mmicko, 2025-04-07:
Main reason is that GHDL is not made to support and hard to adapt to support cross-compile so building it for oss-cad-suite is a problematic.

sigmuha, 2025-04-07:
Okay, so the ghdl plugin is not going to be supported for windows anytime soon?

mmicko, 2025-04-07:
Not soon, since it would require changes in build process of GHDL to make it work and overall lot of time

sigmuha, 2025-04-07:
Ok I see. Thanks for the quick replies!

closed 2025-04-07 by sigmuha