YosysHQ/oss-cad-suite-build #169: build.py script seems to have macOS line endings

buzmeg, 2025-08-19:
It seems like someone checked in build.py with macOS line endings.  I'm on Linux, but wound up with ```/usr/bin/env: ‘python3\r’: No such file or directory```

Git normally recommends that you standardize on Unix line endings in the repo so people can get the correct line endings on checkout.

Thanks.

KrystalDelusion, 2025-08-20:
You appear to be mistaken?
```
$ git ls-files --eol builder.py
i/lf    w/lf    attr/                   builder.py
```

buzmeg, 2025-08-20:
Well, here's my transcript:
```
√ yosys % distrobox enter --root gowin_02
.zshrc being run
√ yosys % git clone https://github.com/YosysHQ/oss-cad-suite-build.git
Cloning into 'oss-cad-suite-build'...
remote: Enumerating objects: 5287, done.
remote: Total 5287 (delta 0), reused 0 (delta 0), pack-reused 5287 (from 2)
Receiving objects: 100% (5287/5287), 1.26 MiB | 7.89 MiB/s, done.
Resolving deltas: 100% (3393/3393), done.
√ yosys % cd oss-cad-suite-build
√ oss-cad-suite-build % ./builder.py build --target=abc
/usr/bin/env: ‘python3\r’: No such file or directory
```

That \r is coming from somewhere.  And I'm on Linux.

Suggestions for extra flags to run that down?

KrystalDelusion, 2025-08-22:
Okay first off, a quick google search suggests that `‘python3\r’` is most likely due to a `crlf`/windows style rather than just a cr _which would require pre OSX macOS_.  Modern macOS uses the same `lf` as linux, and OSX has been around since the early 2000s.

Second, `git ls-files --eol builder.py` lists the index eolinfo as `lf`, i.e. the file _as it exists on github_ uses `lf`, **not** `crlf`.

For contrast, cloning the repo **on windows** and running the same command:
```
git ls-files --eol builder.py
i/lf    w/crlf  attr/                   builder.py
```

Notice how the working directory shows `crlf`, but the index still shows `lf`.  Wherever the \r is coming from, it isn't from github.  Maybe check your global git settings?

buzmeg, 2025-08-22:
This is likely a particular breakage artifact of not running Docker as non-root.  See: #170 

Closing as that's not how this project runs Docker.

closed 2025-08-22 by buzmeg