YosysHQ/oss-cad-suite-build #170: Attempting to build ABC fails with permission denied

buzmeg, 2025-08-19:
Presumably a Docker failure from ABC having been updated recently:

```
==> Building abc for linux-x64 architecture ...
==> Step [ 1/ 1] building abc ...
  -> Remove old output dir ...
  -> Creating output dir ...
  -> Remove old build dir ...
  -> Creating build dir ...
  -> Copy 'abc' source to build dir ...
  -> Compiling ...
bash: /tmp/tmprn708tng: Permission denied
```

Thanks.

KrystalDelusion, 2025-08-20:
OSS Cad Suite is automatically [built nightly](https://github.com/YosysHQ/oss-cad-suite-build/blob/main/.github/workflows/linux-x64.yml), and passed as recently as [40 minutes ago](https://github.com/YosysHQ/oss-cad-suite-build/actions/runs/17086687502).  I'm pretty sure the `tmp/tmp...` failure is on your system, not in the docker.

mmicko, 2025-08-20:
@buzmeg make sure you follow docker setup after installation, since usually something like `sudo usermod -aG docker ${USER}` is required, make sure their helloworld example runs fine as well.

closed 2025-08-22 by buzmeg